FYSETC/FYSETC-S6 #4: Cannot compile a working firmware with TMC_DEBUG enabled

dbemowsk, 2020-01-07:
I am running the S6 board with the Fysetc TMC2209 drivers and every time I enable TMC_DEBUG in configuration_adv.h and upload the firmware to the board, I do not get serial communication from the USB port after rebooting from a firmware upload.  With this disabled I don't get the full diagnostic information when sending an M122 command.  I have posted this problem in multiple places (Fysetc forum, contact form on their website and tried to join their facebook group) and have not received any response from Fysetc tech support from ANY of my posts.  SO now I will try here.

dbemowsk, 2020-01-07:
OK, so I did a test.  In STM32CubeProgrammer there is a checkbox to "run after programming".  I compiled the firmware with TMC_DEBUG enabled and uploaded it to the board with that checked.  Once the upload finishes, DFU disconnects and the computer sees the serial port in device manager.  I connect with a terminal and I can send the M122 command and get the full stats from the drivers.  That's all I tried, but I am sure I could send any valid Gcode and it would probably run just fine.  BUT, the minute I press the reset button or disconnect and reconnect the USB cable, the serial port is lost and not detected again by Windows.  So, I am just making an assumption here that the action of pushing the reset button or plugging in the USB cable is supposed to execute the firmware code, but it doesn't do that when TMC_DEBUG is enabled in the firmware.  That ONLY works if TMC_DEBUG is disabled in the firmware.  I hope that gives someone a clue as to what is going on.

dbemowsk, 2020-01-07:
I ended up finding that the bootloader was corrupted on the board.  After flashing the bootloader again, everything worked.  The part that was confusing an didn't make me think of the bootloader at first was the fact that if I disabled TMC_DEBUG and uploaded, it would appear to work normally.  After uploading the bootloader though all TMC_DEBUG functions worked.

MarkV3082, 2022-04-02:
> I ended up finding that the bootloader was corrupted on the board. After flashing the bootloader again, everything worked. The part that was confusing an didn't make me think of the bootloader at [...]

How did you get the Bootloader to successfully upload to the Board. I have tried using STM32 programmer but I get a can't compile message "Found " when it expected a Hex character on Line 31