FYSETC/FYSETC-S6 #7: SPI port AUX-3

kunpoolsion, 2020-04-19:
Hello,
I wanna add a MAX7219 led matrix to the SPI port close to RGB pins.
Can´t find in definition or scheme what are the corresponding pins in mcu or if those pins are share with other devices.
Can you clarify it?
I´m not gonna use SPI steppers, but I´m using sd reader at exp2.
Thanks

kunpoolsion, 2020-04-19:
Answering myself, it shares pins with EXP1, so basically, can't have a SD card reader on EXP1 and the led matrix at AUX-3

kunpoolsion, 2020-04-19:
There are any other SPI port that can be use?
If yes, what are the corresponding pins for it?

GerogeFu, 2020-09-19:
Sorry , i don't think there are extra pins. This is a little bit old date question , i going to close it , thanks.

closed 2020-09-19 by GerogeFu