FYSETC/FYSETC-S6 #8: 5V Voltage Issue

jdlongenecker, 2020-07-06:
Hello,

I have FYSETC S6 powered by 24V DC Power Supply. Measuring +5V and GND from UART Pin, it reads 4.95-5.0V as expected. However, when I connect Raspberry Pi to USB plug, +5V changes to 4.3-4.4V. This makes my PT100 amplifier inaccurate as it needs 0V and 5V inputs, not 0V and 4.3V input. 

If the control board could use internal 5V2A supply even when USB device is connected, it would work better.

GerogeFu, 2020-09-18:
This has been fix in the new version, thanks for report.

closed 2020-09-18 by GerogeFu

dinamitemic, 2020-09-28:
@GerogeFu Same issue here. Also, if I put the tape on the cable in order to do not share the 5V it is not detected anymore. I think it's missing the pullup.
How we can solve this?