FYSETC/FYSETC-S6 #10: I2C devices not working ...

kbahey, 2020-08-27:
I am using the S6 board for a telescope project (OnStep).
It works, and controls the scope as expected. 

The issue is that connecting I2C devices (e.g. BME280 for ambient temperature, pressure and humidity) does not work at all.

GerogeFu, 2020-09-18:
If you are using the EEPROM I2C port for you telescope, you need to take care of the extra circuit of this port, you can look into the schematic.

kbahey, 2020-09-18:
The issue is that connecting any I2C device to the I2C pins on the S6 does not work. 
We tried the BME280 as an example, with proper VCC and GND to it. 
It does not work. 

Many people tried, and no one succeeded.

GerogeFu, 2020-09-19:
This IIC circuit is originally for EEPROM , there are extra resistor/cap in the circuit , maybe you need to avoid them
[image]