FYSETC/FYSETC-S6 #11: Cannot force DFU mode with GPS powered up

kbahey, 2020-08-27:
I am using the S6 board for a telescope project (OnStep).
It works normally for most functions.

However, when having the GPS connected, the board would not enter DFU mode when the BOOT0 button is pressed, then USB cable is inserted. 

If I disconnect the 5V going to the GPS, then DFU mode works normally. 

Any idea how to solve this?

GerogeFu, 2020-09-18:
I think your power for the GPS goes to back to S6 , so you can power the GPS with S6 but not extra power supply.

kbahey, 2020-09-23:
Can you explain your reply? I don't understand it.

GerogeFu, 2020-09-24:
Oh , ok . I mean your GPS module is feeding the S6 board. But what we need is  S6 board feeding the GPS.

kbahey, 2020-09-24:
No.
The GPS module does not provide power to the S6 board.
It takes power from the board.
And the board takes power from the USB cable.

When I use 12V input to the board, and then press the Boot0, then the Reset button, the board sometimes goes into DFU mode, but sometimes it does. Better than USB only, but not consistent.