FYSETC/FYSETC-S6 #12: [HINT] Problem with TMC and missing vmot

GMagician, 2020-09-09:
I know that 2.0 board is on development and I know that some TMC drivers have a lot of issues (they die) when vmot is missing but vin is present. To solve this issue it's a common way to use vin with a diode to power vmot.

I don't know if 2.0 may be powered (via usb?) with no 12/24V. In such case TMC lose their cnf and requires a reset + reprogram when 12v/24v is given (I noticed this acting at reverse, tmc fully programmed and removed 24V, once 24V comes back tmc 2209 registers are wrong).

There  are some issues reported in Marlin about this weird behavior. On a different test board (ramps) I solved this powering TMC vmot with 5V + diode. It seems that 5v instead of 3.3 solve the issue.

Since I'm looking for 2.0 board it would be a good thing if it may solve such issue

Thanks for your attention

GerogeFu, 2020-09-18:
Thanks very much for this report. I will discuss this with the team. Thanks.

GerogeFu, 2020-09-19:
Discussed with the team, our 2.0 is going to market, so the board is unchangeable, we will try to fix it in 2.1 version. All regards.

closed 2020-09-19 by GerogeFu

GMagician, 2020-09-19:
Ok, I'll wait for 2.1 then.

Thanks for considering my hint.

To complete it, the TMC register I noticed to change were the ones involved in sensorless homing (I'm almost sure current limit was one of them)

GerogeFu, 2020-09-19:
Ok , i want to know if user power the stepper driver with the power supply , this won't bother them, right ? It only happened when stepper driver power by USB/5v, following 12v/24v.

GMagician, 2020-09-19:
This is how to replicate issue:

USB + 24V (no matter order) and home: everything ok
removed 24V: TMC registers changed.
reapplyed 24V: TMC register still changed
home: sensorless is done with too high current (and I almost sure it's not the only register changed)

GMagician, 2020-09-19:
you may read https://github.com/MarlinFirmware/Marlin/issues/18498 to see how registers change, there is a Marlin log

GMagician, 2020-09-19:
here is a syummary of what is changed after power off
before:
15:05:37.418 : Set current 900 900 900 1000 250
15:05:37.441 : RMS current 887 887 887 994 244
15:05:37.503 : CS actual 14/31 14/31 14/31 8/31 3/31
15:05:37.525 : PWM scale 16 16 16 10 4
15:05:37.556 : vsense 1=.18 1=.18 1=.18 0=.325 1=.18
15:05:37.603 : msteps 16 16 16 16 16
15:05:37.688 : off time 4 4 4 4 4
15:05:37.719 : blank time 24 24 24 24 24
15:05:37.741 : -end 2 2 2 2 2
15:05:37.773 : -start 1 1 1 1 1
15:05:38.058 : Driver registers:
15:05:38.073 : X 0xC0:0E:00:00
15:05:38.073 : Y 0xC0:0E:00:00
15:05:38.089 : Z 0xC0:0E:00:00
15:05:38.089 : Z2 0xC0:08:00:00
15:05:38.089 : E 0xC0:03:00:00
after: 
5:06:57.700 : RMS current 1602 1602 1602 994 441
15:06:57.738 : MAX current 2259 2259 2259 1402 622
15:06:57.769 : CS actual 31/31 31/31 31/31 31/31 31/31
15:06:57.784 : PWM scale 36 36 36 36 36
15:06:57.816 : vsense 0=.325 0=.325 0=.325 0=.325 0=.325
15:06:57.869 : msteps 8 8 8 8 8
15:06:57.954 : off time 3 3 3 3 3
15:06:57.985 : blank time 36 36 36 36 36
15:06:58.016 : -end -3 -3 -3 -3 -3
15:06:58.038 : -start 6 6 6 6 6
15:06:58.333 : Driver registers:
15:06:58.339 : X 0xC0:1F:00:00
15:06:58.339 : Y 0xC0:1F:00:00
15:06:58.355 : Z 0xC0:1F:00:00
15:06:58.355 : Z2 0xC0:1F:00:00
15:06:58.370 : E 0xC0:1F:00:00

as you can see are a lot of values

GerogeFu, 2020-09-21:
Thanks for your report. I get it.

GMagician, 2020-11-04:
@GerogeFu sorry to bother, I see that in V2.0 there is no vmot diode protection at all but you can power using usb 5V. I'm expecting that doing such option will kill a lot of tmc drivers. Do you think you will add such diode in new board revisions? Do you think such diode will work on 5V instead 3.3V?

Thanks