FYSETC/FYSETC-S6 #13: Updated info on S6 v2.0 wanted

Bajotaz, 2020-09-24:
I am unable to find any info regarding v2.0.

John-SC077, 2020-09-27:
I also need some help with this release.

I received 3 V2.0 boards yesterday, but cannot move my code to them because the code cannot operate the TMC 5160 drivers on this version.  The same code is operating on V1.2 boards without issue, but the TMC 5160 drivers do not energize the stepper motors when on the V2.0 S6 board.  The remainder of the sketch works perfectly including running the WiFi on the UART port and a connection to another Arduino device on Serial3 (on the EXP1 connector.

GerogeFu, 2020-09-27:
I update the README of S6 , try to follow it.

Bajotaz, 2020-09-27:
> I also need some help with this release.
>
> I received 3 V2.0 boards yesterday, but cannot move my code to them because the code cannot operate the TMC 5160 drivers on this version. The same code is [...]

//
// Steppers
//
#define X_ENABLE_PIN                        PE9

#if HAS_TMC_UART
  #define X_SERIAL_TX_PIN                   PE8
  #define Y_SERIAL_TX_PIN                   PC4
  #define Y_SERIAL_RX_PIN                   PC4
  #define Z_SERIAL_TX_PIN                   PD12
  #define E0_SERIAL_TX_PIN                  PA15
  #define E1_SERIAL_TX_PIN                  PC5
  #define E2_SERIAL_TX_PIN                  PE0
#endif

Bajotaz, 2020-09-27:
> I update the README of S6 , try to follow it.

Thanks, found my info now.

closed 2020-10-08 by GerogeFu