FYSETC/FYSETC-S6 #16: "Cut Jumpers" for 5V AD597K

Jhonno84, 2021-04-08:
Can someone explain in very simple terms how to "cut the jumper" so I can run a PT100 through an AD597K board?

Marconet68, 2021-04-11:
I've put some explanation on "FYSETC" facebook group: [https://www.facebook.com/groups/197476557529090/permalink/812555469354526/?comment_id=814816759128397&notif_id=1618073626372899&ref=notif&notif_t=group_comment_mention](url).
It will very helpfull that somebody from Fysetc confirm the connection.

But AD597K is for Thermocouple type K not for PT100....
[image]

Jhonno84, 2021-04-11:
Hi Marco, 

Thanks for taking time with me on this.. 

I seem to have made an incorrect assumption.. When I searched for the AD597 only AD597K came up. So I thought it was the same thing. I have ordered a MAX31865 also..

Jhonno84, 2021-04-30:
I can confirm the above works. I am running an AD597K board, with a PT100, jumper cut and board powered off an alternative 5V supply. Running Klipper, you need to configure the board to 3.3v not 5v as PC0 is 3.3V.