FYSETC/FYSETC-S6 #19: New bootloader

GMagician, 2021-07-08:
Hi, not sure this is the right place.
I tryed to upload new bootloader 0x8000 with stcube.
Once loaded, always with dfu and VScode I uploaded Marlin S6_8000 env but once I reboot (boot jumper and power off/on) no usb port happears.

Since it doesn't work I re uploaded boot_10000 and marlin s6 env and it works.

What's wrong?

GMagician, 2021-07-08:
Precompiled Marlin, loaded with stcube, works. Only the source with vscode seems to hang.
I wrote to Marlin github

GerogeFu, 2021-07-09:
Did you try the branch in this repo, and what upstream Marlin are you using? The right one should be bugfix-2.0.x.

GMagician, 2021-07-09:
Marlin in this repo is working, I tryed the very last bugfix in Marlin, compiled for 8000 env, that seems not to work

GerogeFu, 2021-07-12:
Did you raise an issue for Marlin ?

GMagician, 2021-07-12:
I wrote on PR which merged 8000 version, but I think it's good to open an issue

kyon03, 2021-07-13:
same issue whats the solution?

rsdcampos, 2021-07-21:
Same issue for me to. If I use the 0x10000 the eeprom don't save the changes, when I do a M500 it returns "Error writing to EEPROM" and EEPROM datasize error". There are any solution for it?

GerogeFu, 2021-07-22:
Check the PR here https://github.com/MarlinFirmware/Marlin/pull/22421

rsdcampos, 2021-07-22:
> 
> 
> Check the PR here [MarlinFirmware/Marlin#22421](https://github.com/MarlinFirmware/Marlin/pull/22421)

Nope, I made your changes in Marlin bugfix and used your entire commit using my configurations files without sucess in both, same issue. Just for complete answer, yes, I changed the envs in platformio.ini and tryed both variants without sucess.

GerogeFu, 2021-07-31:
Now should work. Check the [PR](MarlinFirmware/Marlin#22421) there.

GMagician, 2021-07-31:
Yep that solve this

closed 2021-07-31 by GMagician