FYSETC/FYSETC-S6 #20: Clipper and new bootloader

asha77, 2021-07-26:
Hi, 
i've got error with new bootloader and klipper firmware with S6 board. 
Firmware uploaded using STM32cube and DFU.
Sequence:
1. Enter DFU Mode
2. Upload Latest bootloader 
3. Restart Board 
4. Upload latest klipper compiled firmware. Start address set to 0x08010000
5. Flashing successfull
6. Attach S6 to RPI 
7. Starting Klipper, Firmware working OK, I can print
8. When I reboot board with BOOT jumper set to DFU - it goes to DFU mode (expected)
9. Whet I remove DFU Jumper and reboot board - USB cannot detect board anymore. It becomes brick.

I can repeat steps 1-7 to make it work again, but it is not expected scenario. 

Please, advice what to do with that, thank you in advance.

GerogeFu, 2021-07-30:
Why you want to set to DFU at step 8?

asha77, 2021-07-30:
Just to check how it starting with and without jumper. 
Important note: on step 7 it starts OK from STM32cube, not from hard reset (power cycle reset button). 
Steps 8, 9 - made with hard reset.

GerogeFu, 2021-07-31:
You can use our prebuild to test the bootloader: https://github.com/FYSETC/FYSETC-S6/tree/main/firmware/Pre-builds, it is Marlin build, you can send "M105" command to test if board works ok. I double check my side,  it works. And i recommend a hard reset after you flash firmware.

teeedubb, 2022-04-15:
Hi,

I am experiencing the same scenario exactly. Did you ever resolve this?

asha77, 2022-04-15:
Hi, after some manipulations i get board back to work. 
Unfortunately i not tried to reproduce this issue, i can't privide any clear instruction to fix it.

teeedubb, 2022-04-15:
Compiling klipper for no bootloader solved the issue for me, it now works after a reboot.

asha77, 2022-07-13:
I got this issue once again. yes, klipper with no bootloader option did it:
- compile klipper.bin with no bootloader
- set BOOT jumper to DFU mode
- ensure DFU mode active: dfu-util -l -v 
- flash firmware: dfu-util -d 0483:df11 -a 0 -R -D  klipper_no_bootloader.bin -s0x08000000

After restart S6 octoprint can communicate with klipper.