FYSETC/FYSETC-S6 #21: Missing pinout drawing file

rsdcampos, 2021-08-01:
I like a lot this board, but a thing I miss a lot to use it is a drawing with the correspondent pinout in place, so I wold like to know if you could provide some good stuff in this way.
some like this:
https://github.com/FYSETC/FYSETC-F6/blob/master/images/F6V1.3-PIN.svg
or this:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/raw/master/hardware/MKS%20Robin%20E3%20V1.0_003/MKS%20Robin%20E3%20V1.0_003%20PIN.pdf
or this:
https://user-images.githubusercontent.com/12979070/68524259-9e103600-02ff-11ea-8348-4f895cb3e100.png

Thanks a lot.

GerogeFu, 2021-08-02:
See, we will add one. You can check the pin at the back of the board first. I will update here if we done.

rsdcampos, 2021-08-02:
Thanks for the fast answer. Not all the pins are available at back of board, for this reason I ask for pinout. For example, I can't know the order of the out pins for the motors, I need to try and if the motor runs inverted I invert the cable, if there are a pin information I could plug without error. This is just a Example, there are many other good reasons to have a good pinout picture or pdf available here.

Thanks.

teeedubb, 2022-03-27:
Hi @GerogeFu,

Any updates to this?

rsdcampos, 2023-02-01:
Hi, I saw the new pinout file and this is the best pinout file I ever seem. Congratulations and thanks to consider my comments.

closed 2023-02-01 by rsdcampos