FYSETC/FYSETC-S6 #35: S6 Missing klipper bootloader firmware for flashing MCU over Uart

Max-gee, 2022-05-11:
FYSETC S6 is proving a great board for klipper. and in conjunction with a RPI Z2 and Fluid/Mainsail is almost a perfect setup for most people to use or upgrade to.

the only issue thing that stops me recommending this as 100% perfect is that its not currently easy to upgrade the MCU firmware when klipper is updated without disconnecting the board from UART and rejumpering to setup for USB.

Section 3.3.2 of FYSETC's own wiki states this should be possible 
_The UART socket next to the limit switch can be used to connect serial devices such as the wifi module, or **to update the firmware through the serial port.**_

Recently FYSETC released a bootloader compiled for the similar Spider board which uses the same MCU as the S6. 

therefore quick questions :-
1. Can anyone tell me if a suitable equivalent is available for the S6 that i'm missing or havnt found 
2. could I load the Spiders bootloader firmware  (klipper-32k-Uart.bin) on the S6 
3. How involved would it be to compile a suitable bootloader for the S6 that enables serial uart flashing 
and finally is there any other way to flash over the RPI serial UART that would work without needing a special bootloader ?

teeedubb, 2022-05-14:
I too am interested in this. Disappointed that the wiki states that it is possible, but I have not found a way to do so.

Max-gee, 2022-10-26:
> Is there a solution to this problem?
> I have also tried but without hope
> I think that FYSECT will not update this card anymore.
> So it is obsolete for FYSECT.
>
> There have been other problems with this card and FYSECT has done nothing.
> I am thinking of getting a new card but not from this brand who do not
> provide any support to their customers.

No solution but it's not on fysetc, after researching it's clear that
almost all MCU's cant be flashed over uart as they needs a special " data
stream" sent over USB to signal the MCU to prepare to go into flashing mode.

Over USB that's doable but not serial over uart